jameszah/ESP32-CAM-Video-Recorder-junior #1: Re-adding ftp server functionality

jach999, 2020-11-10:
First of all I wanted to thank you for your work, I really admire what you have done with the programming of this module.
Secondly, I must say that I am not a programmer, I am simply a bungler who copies and pastes code praying not to break it!
I have a project in which I want my camera to be activated as quickly as possible through a motion detector (RCWL-0516) to observe insects that fly past a device. Naturally, I also need it to be at the maximum fps level, the video quality does not matter so much. Thanks to your ESP32-CAM-Video-Recorder-junior code those two conditions can be achieved surprisingly well. However, I found it a shame that the ftp server features that are in the master code are not available. So I have tried to copy and paste again the code that enables this function. It has worked miraculously, however I think this has caused the streaming function to crash indefinitely after a few seconds of working.
I would be extremely grateful if you can review my code and see if you can find the error!
Thank you again for your developments!

[ESP32-CAM-Video-Recorder-junior-07x_FTP.zip](https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/files/5518262/ESP32-CAM-Video-Recorder-junior-07x_FTP.zip)

jameszah, 2020-11-10:
You need to send the localip string a couple more time to fill in the http text and url of the ftp server ... in the index_handler function. (see photo below)

Nice little mod! 😄 

It is amusing for the version control police that the strange file names "desklens 1.5 + 3912s.avi" and the lack of dates in the ftp program show that you got the avi writer from 2-3 version ago, and the ftp software from about 10 versions ago. 😄 

[image]

jameszah, 2020-11-11:
You could get rid of the delay(10) at the end and set the quality to 7 or 8 (from 20) to improve the picture, without reducing the recording speed -- but getting bigger files.

jameszah, 2020-11-11:
I commented out these lines -- which don't seem to do anything useful -- might cause problems with the camera.  It was running here on my desk and would stop streaming sometimes. ???

```
    res = httpd_resp_send_chunk(req, _STREAM_BOUNDARY, strlen(_STREAM_BOUNDARY));
    if (res != ESP_OK) {
      return res;
    }

    //if (fb_next == NULL) {
    //  esp_camera_fb_return(fb);
    //}

    delay(200);       // 200 ms = 5 fps !!!
```

jach999, 2020-11-11:
many thanks for your reply!! Your right, the code is a Frankenstein but... it's alive!!! 😄
now it works fine without interruption while streaming!! Great!! Thank you again!!!

al3xv0lk, 2021-02-16:
I'm testing as well with ftp on Junior, but it is not working, I even used the code of @jach999 with @jameszah tips, but it is not working at all, I tested in my android and pc, on FileZilla I'm getting this:
`Status:	Connecting to 192.168.1.235:21...
Status:	Connection established, waiting for welcome message...
Response:	220--- Welcome to FTP for ESP8266 ---
Response:	220---   By David Paiva   ---
Response:	220 --   Version FTP-2016-01-14   --
Command:	AUTH TLS
Response:	500 Syntax error
Command:	AUTH SSL
Response:	530 user not found
Status:	Insecure server, it does not support FTP over TLS.
Command:	USER esp
Error:	Could not connect to server
`

jameszah, 2021-02-21:
I think you need to shut off security - the SSL and TLS stuff.

Username and password "esp".

[image]

[image]

jameszah, 2022-01-13:
... new version better than ftp!

closed 2022-01-13 by jameszah