jameszah/ESP32-CAM-Video-Recorder-junior #2: How to make timelapse movie?

AnthonyKNorman, 2020-12-21:
Hi
You mention:
at SVGA 25fps, quality 12 in the sunshine, you get files of 1.5 GB per 30 minutes - so reduce quality, switch to lower framesize, or go to the timelapse system of 1 frame per second or 10 frames per second, etc to get smaller files
How do I "go to the timelapse system"?
Thanks

jameszah, 2020-12-21:
https://github.com/jameszah/ESP32-CAM-Video-Recorder

The other program has timelapse where you set the interval between frames in the settings.h file, or at the http request

int  capture_interval = 10000;       //  milli-seconds between frames

http://desklens.local/start?framesize=VGA&length=1800&interval=250&quality=10&repeat=100&speed=1&gray=0&pir=0&bot=1

This junior version runs at full speed of camera and sd card -- so you could probably just add a delay(1000) inside the loop, and it will slow down to 1 frame per second.

jameszah, 2020-12-26:
More info here
https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/issues/3#issuecomment-751402651

closed 2022-01-13 by jameszah