jameszah/ESP32-CAM-Video-Recorder-junior #3: Setting Framerate of Recording?

sunny52266, 2020-12-26:


jameszah, 2020-12-26:
So you want to use the junior program, but run it slower than full-speed?

I think this is the same question as here https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/issues/2#issuecomment-749071736

I haven't tried this yet, but you could just add a single line at the top of the loop to delay() until the next frame.
So if you want 10 frames per second or 0.1 seconds per frame or 100 milli-seconds between frames, then just put a delay(100) at the top of the loop.

On a SVGA movie, the camera takes about 40 milli-seconds, and the SD write is a little quicker, but the camera is taking the picture while your loop() is in the delay, but the SD write does take time from the loop().

So a delay(100) at the top of every loop will slow you down to 10 frames per second.  And the "take picture" time will be less than 1 millisecocond (because it is already done), plus the 20 milli-seconds (?) for the SD write, then you will be at about 8.3 frames per second (1000 / (100 + 20)) 

SO you could measure the SD write speed and lower the delay to 80 or so, for 10 fps.

If it is 10 seconds per frame - delay(10000) - the the SD write time isn't too relevant.   

I must try this before I suggest it again 😄

jameszah, 2020-12-26:
ok, I tried it -- and it worked.

delay(180) at the top of loop() delivers 5 fps, with about 16 milliseconds to write the data to SD (I was guessing 20).  That will depend on the rich colors and complexity of your scene.

```
11:24:59.039 -> Recorded 120s in 614 frames
11:24:59.039 -> File size is 12168432 bytes
11:24:59.039 -> Actual FPS is 5.11
11:24:59.039 -> Max data rate is 99031 byte/s
11:24:59.039 -> Frame duration is 195767 us
11:24:59.039 -> Average frame length is 19804 bytes
11:24:59.039 -> Average picture time (ms) 0.00
11:24:59.073 -> Average write time (ms)   15.98
11:24:59.073 -> Normal jpg % 93.0
11:24:59.073 -> Extend jpg % 6.8
11:24:59.073 -> Bad    jpg % 0.00000
11:24:59.073 -> Writng the index, 614 frames
11:24:59.140 -> File open: /sdcard/idx.tmp
11:24:59.240 -> ---
11:24:59.240 -> End the avi at 174642.  It was 614 frames, 174642 ms at 5.0 fps...
```

jameszah, 2020-12-26:
So, if you slow the recording to 5 fps, it will play back at realtime and 5 fps on your computer.
To speed up the playback, your can have your computer do that, or change line 815 of junior-10x.ino to:

    uint8_t iAttainedFPS = round(fRealFPS) * 5;

... which will tell the computer to speed up 5 times, and you will get 25 fps for smooth playback in timelapse.

So for 5 frames per second recording, playing back at 5 times speed, or 25 frames per second
```
line 814   uint8_t iAttainedFPS = round(fRealFPS) * 5;
line 1325 delay (180);

```

or for 1 seconds per frame recording , playback at 30 times, or 30 fps, or 30 minutes played in 60 seconds
```
line 814   uint8_t iAttainedFPS = round(fRealFPS) * 30;
line 1325 delay (1000);
```

sunny52266, 2020-12-28:
Great Mate! Thanks for your work! the suggestion works!

closed 2020-12-28 by sunny52266