jameszah/ESP32-CAM-Video-Recorder-junior #4: Can ESP32-CAM-Video-Recorder WifiManeger be added?

thenorhstar, 2021-01-01:
Hı,
First of all, thank you very much for the application.

I tried to add ESP32-CAM-Video-Recorder-junior-10x WifiManager but it gives an error in the library (#include "esp_http_server.h").

```

//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
// Time
#include "time.h"
#include <WiFi.h>
#include <WiFiManager.h>  // **I added here**
#include <ESPmDNS.h>

time_t now;
struct tm timeinfo;
char localip[20];

bool init_wifi()
```
It gives the following errors. 

 If you do not mind, can you add WifiManager feature?

```
//**************
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h:30:0,
                 from D:\Dokumanlarim\Ardunio\arduino-nightly-windows\Arduino_taslak\libraries\WiFiManager/WiFiManager.h:70,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:908:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/HTTP_Method.h:5:18: error: redeclaration of 'HTTP_GET'
   HTTP_GET     = 0b00000001,
                  ^
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_server/esp_http_server.h:22:0,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:133:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:138:31: note: previous declaration 'http_method HTTP_GET'
 #define XX(num, name, string) HTTP_##name = num,
                               ^
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:96:3: note: in expansion of macro 'XX'
   XX(1,  GET,         GET)          \
   ^
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h:30:0,
                 from D:\Dokumanlarim\Ardunio\arduino-nightly-windows\Arduino_taslak\libraries\WiFiManager/WiFiManager.h:70,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:908:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/HTTP_Method.h:6:18: error: redeclaration of 'HTTP_POST'
   HTTP_POST    = 0b00000010,
                  ^
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_server/esp_http_server.h:22:0,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:133:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:138:31: note: previous declaration 'http_method HTTP_POST'
 #define XX(num, name, string) HTTP_##name = num,
                               ^
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:98:3: note: in expansion of macro 'XX'
   XX(3,  POST,        POST)         \
   ^
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h:30:0,
                 from D:\Dokumanlarim\Ardunio\arduino-nightly-windows\Arduino_taslak\libraries\WiFiManager/WiFiManager.h:70,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:908:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/HTTP_Method.h:7:18: error: redeclaration of 'HTTP_DELETE'
   HTTP_DELETE  = 0b00000100,
                  ^
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_server/esp_http_server.h:22:0,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:133:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:138:31: note: previous declaration 'http_method HTTP_DELETE'
 #define XX(num, name, string) HTTP_##name = num,
[... 53 lines cut ...]
   XX(2,  HEAD,        HEAD)         \
   ^
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h:30:0,
                 from D:\Dokumanlarim\Ardunio\arduino-nightly-windows\Arduino_taslak\libraries\WiFiManager/WiFiManager.h:70,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:908:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/HTTP_Method.h:11:18: error: redeclaration of 'HTTP_OPTIONS'
   HTTP_OPTIONS = 0b01000000,
                  ^
In file included from C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_server/esp_http_server.h:22:0,
                 from C:\Users\Home_PC\Desktop\ESP32-CAM-Video-Recorder-master(1)\ESP32-CAM-V10\ESP32-CAM-V10x\ESP32-CAM-V10x.ino:133:
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:138:31: note: previous declaration 'http_method HTTP_OPTIONS'
 #define XX(num, name, string) HTTP_##name = num,
                               ^
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:102:3: note: in expansion of macro 'XX'
   XX(6,  OPTIONS,     OPTIONS)      \
   ^
C:\Users\Home_PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^

//*************
```

jameszah, 2021-01-03:
It is a somewhat difficult problem.
There is a problem with WiFiManager (using WebServer.h) and my code (using esp_http_server.h) both trying to own the http interface, and also both of them want to own the HTTP_GET symbol.

There is a solution over here -- using IOTWeb??, a similar thing to WiFiManager. but it is not very nice solution.

https://github.com/jameszah/ESP32-CAM-Video-Recorder/issues/35#issuecomment-742811433

I'll take another try at it  next week.

thenorhstar, 2021-01-04:
Thank you for your attention. We are looking forward.
Can the date and time be added on the video recording? Is this possible?

jameszah, 2021-01-04:
ok, I think I got it.

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/master/ESP32-CAM-Video-Recorder-junior-10x.wifiman.ino

You use the normal WiFiManager procedure 192.168.4.1 to set the ssid and password, and then you have to reboot the esp32 to let it start the web services correctly.

I have not tried it outside of my wifi zone yet - I assume it will try to start wifi, then try to start config, and then fail, and run the program as a simple movie camera.  There are some commented lines in the code if it doesn't timeout, or takes too long to timeout.

closed 2022-01-13 by jameszah