jameszah/ESP32-CAM-Video-Recorder-junior #5: v09 won't compile

andrewherron, 2021-01-08:
I'm trying to compile v09 with and I'm getting the following errors:

`Arduino: 1.8.13 (Mac OS X), Board: "AI Thinker ESP32-CAM"

/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'void print_quartet(long unsigned int, FILE*)':
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:294:10: warning: unused variable 'i1_err' [-Wunused-variable]
   size_t i1_err = fwrite(y , 1, 4, fd);
          ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'void print_2quartet(long unsigned int, long unsigned int, FILE*)':
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:311:10: warning: unused variable 'i1_err' [-Wunused-variable]
   size_t i1_err = fwrite(y , 1, 8, fd);
          ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'esp_err_t config_camera()':
ESP32-CAM-Video-Recorder-junior-10x:433:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'esp_err_t init_sdcard()':
ESP32-CAM-Video-Recorder-junior-10x:466:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'esp_err_t start_avi()':
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:618:10: warning: variable 'err' set but not used [-Wunused-but-set-variable]
   size_t err = fwrite(buf, 1, AVIOFFSET, avifile);
          ^
ESP32-CAM-Video-Recorder-junior-10x:698:1: error: no return statement in function returning non-void [-Werror=return-type]
 } // end of start avi
 ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'esp_err_t another_save_avi(camera_fb_t*)':
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:721:8: warning: unused variable 'frame_write_start' [-Wunused-variable]
   long frame_write_start = millis();
        ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:776:8: warning: unused variable 'frame_write_end' [-Wunused-variable]
   long frame_write_end = millis();
        ^
ESP32-CAM-Video-Recorder-junior-10x:786:1: error: no return statement in function returning non-void [-Werror=return-type]
 } // end of another_pic_avi
 ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'esp_err_t end_avi()':
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:806:9: warning: unused variable 'xx' [-Wunused-variable]
     int xx = remove("/sdcard/idx.tmp");
         ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:807:9: warning: unused variable 'yy' [-Wunused-variable]
     int yy = remove(fname);
         ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:885:14: warning: unused variable 'res' [-Wunused-variable]
       size_t res = fread ( AteBytes, 1, 8, idxfile);
              ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:886:14: warning: unused variable 'i1_err' [-Wunused-variable]
       size_t i1_err = fwrite(dc_buf, 1, 4, avifile);
              ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:887:14: warning: unused variable 'i2_err' [-Wunused-variable]
       size_t i2_err = fwrite(zero_buf, 1, 4, avifile);
              ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:888:14: warning: unused variable 'i3_err' [-Wunused-variable]
       size_t i3_err = fwrite(AteBytes, 1, 8, avifile);
              ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:868:12: warning: unused variable 'i1_err' [-Wunused-variable]
     size_t i1_err = fwrite(idx1_buf, 1, 4, avifile);
            ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:895:9: warning: unused variable 'xx' [-Wunused-variable]
     int xx = remove("/sdcard/idx.tmp");
         ^
ESP32-CAM-Video-Recorder-junior-10x:899:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'esp_err_t index_handler(httpd_req_t*)':
ESP32-CAM-Video-Recorder-junior-10x:1057:45: error: format '%d' expects argument of type 'int', but argument 9 has type 'long int' [-Werror=format=]
           localip, localip, localip, localip);
                                             ^
ESP32-CAM-Video-Recorder-junior-10x:1057:45: error: format '%d' expects argument of type 'int', but argument 9 has type 'long int' [-Werror=format=]
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:1013:14: warning: unused variable 'the_message' [-Wunused-variable]
   const char the_message[] = "Status";
              ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'esp_err_t photos_handler(httpd_req_t*)':
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:1072:14: warning: unused variable 'the_message' [-Wunused-variable]
   const char the_message[] = "Status";
              ^
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: In function 'void loop()':
ESP32-CAM-Video-Recorder-junior-10x:1413:62: error: format '%d' expects argument of type 'int', but argument 3 has type 'long int' [-Werror=format=]
     Serial.printf("Start the avi ... at %d\n", avi_start_time);
                                                              ^
ESP32-CAM-Video-Recorder-junior-10x:1446:152: error: format '%d' expects argument of type 'int', but argument 3 has type 'long unsigned int' [-Werror=format=]
     Serial.printf("End the avi at %d.  It was %d frames, %d ms at %.1f fps...\n", millis(), frame_cnt, avi_end_time, avi_end_time - avi_start_time, fps);
                                                                                                                                                        ^
ESP32-CAM-Video-Recorder-junior-10x:1446:152: error: format '%d' expects argument of type 'int', but argument 5 has type 'long int' [-Werror=format=]
ESP32-CAM-Video-Recorder-junior-10x:1446:152: error: format '%f' expects argument of type 'double', but argument 6 has type 'long int' [-Werror=format=]
ESP32-CAM-Video-Recorder-junior-10x:1446:152: error: too many arguments for format [-Werror=format-extra-args]
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino: At global scope:
/ESP32-CAM-Video-Recorder-junior/ESP32-CAM-Video-Recorder-junior-10x/ESP32-CAM-Video-Recorder-junior-10x.ino:166:18: warning: 'cam_err' defined but not used [-Wunused-variable]
 static esp_err_t cam_err;
                  ^
cc1plus: some warnings being treated as errors
Multiple libraries were found for "WiFi.h"
 Used: Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: Documents/Arduino/libraries/WiFi
exit status 1
no return statement in function returning non-void [-Werror=return-type]
`
`

jameszah, 2021-01-08:
What compiler ide are you using?
I see Moc OS, and then a hundred warnings about unused variables, and complaints about "int" vs "long int".
Not clear which is the actual problem, but it is much more aggressive error checking a Arduino IDE.
Are some of these lines in red?  Or can you turn off the relentless error checking?

andrewherron, 2021-01-08:
Ah, that was it, thanks and sorry for the alarm. I had never had that happen before; the latest update to the Arduino update had it checking all warnings as errors instead of it's normal behavior.

closed 2021-01-08 by andrewherron