jameszah/ESP32-CAM-Video-Recorder-junior #8: esp_camera_fb_get(): Failed to get the frame on time!

shyney7, 2021-03-28:
Hi I keep getting this error and the board keeps rebooting.

Framework Version: 1.0.4
sd card format:        FAT32

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

---
                                    
-------------------------------------
ESP32-CAM-Video-Recorder-junior v10
-------------------------------------
setup, core 1, priority = 1
Starting the wifi ...
.
Internet status: 3
mDNS responder started 'desklens'
Local time: Thu Jan  1 01:00:35 1970
IP: 192.168.1.103
 
Setting up the camera ...

Camera started correctly, Type is 26 (hex) of 9650, 7725, 2640, 3660, 5640

[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
SD card mount successfully!
Name: SL32G
Type: SDHC/SDXC
Speed: 40 MHz
Size: 29664MB
Warming up the camera ... here are some frames sizes ...
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2078  PS      : 0x00060930  A0      : 0x800d572e  A1      : 0x3ffb1f00  
A2      : 0x3ffd2d64  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f401db2  
A6      : 0x00000000  A7      : 0x3ffba264  A8      : 0x800d2072  A9      : 0x3ffb1ed0  
A10     : 0x00000000  A11     : 0x3f401db2  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffd4040  A15     : 0x3f401c96  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d2078:0x3ffb1f00 0x400d572b:0x3ffb1fb0 0x4008ee9d:0x3ffb1fd0

Rebooting...
```

This is the board:
[image]

I hope you can help me.

jameszah, 2021-03-28:
Looks like you can talk to camera because it knows it is type 26, but the frame not arriving, so look at these frame size parameters.  Make framesize small (vga) and smaller than the config size (uxga), and lower the quality (to 20), and make sure you have good power, and everything is plugged in nicely.  You could try the simple test program to take a picture before you try to record.

It is just rebooting because the camera isn't working.

  config.xclk_freq_hz = 10000000;     // 10000000 or 20000000 -- 10 is faster !!
  config.pixel_format = PIXFORMAT_JPEG;

~~~~~

int c1_framesize = 7;                //  10 UXGA, 9 SXGA, 7 SVGA, 6 VGA, 5 CIF
int c1_quality = 10;                 //  quality on the 1..63 scale  - lower is better quality and bigger files - must be higher than the jpeg_quality in camera_config
int c1_avi_length = 1800;     

~~~~~~~

  config.frame_size = FRAMESIZE_UXGA; // edit in framesizes below
  config.jpeg_quality = 6;  // 1 to 63 - smaller number is higher quality and more data - must be lower rhat the quality parameter at the top
  config.fb_count = 7;

shyney7, 2021-03-28:
@jameszah 
Thank you for the quick reply. Unfortunately even after changing to framesize 6 or 5 (VGA and CIF) I keep getting the same error message.

I tried using a sketch that only boots and takes a picture and saves it to the sd-card and then goes to sleep. 
Source: https://raw.githubusercontent.com/RuiSantosdotme/ESP32-CAM-Arduino-IDE/master/ESP32-CAM-Take-Photo-Save-MicroSD-Card/ESP32-CAM-Take-Photo-Save-MicroSD-Card.ino

Also there I tried changing the quality to CIF with:
```
if(psramFound()){
    config.frame_size = FRAMESIZE_CIF; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
    config.jpeg_quality = 15;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_CIF;
    config.jpeg_quality = 20;
    config.fb_count = 1;
  }
```
But I'm still getting the error:
```
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
Camera capture failed
```

jameszah, 2021-03-28:
I'm guessing wires and electrical issues.  I assume you have 5 volts coming from underneath.  Use short thick wires and use a power supply rather than the ftdi power through a usb to a usb port 6 feet away.  Take the camera off and put it back again.  Try a different camera, or a different esp32 module.  There are message groups with hundreds of messages on that error ... and mostly the solution is power, wires, bad camera, ...  

Or pull it off breadboard, and just run the wires directly to module.

shyney7, 2021-03-28:
@jameszah 
It was really as simple as unplugging the camera and putting it back on. I would have never tought of such a simple solution. :)

I wasnt powering it up with the FTDI Programmer I soldered a custom connector from an old USB Cable. I dont like to use the power from the FTDI Programmer anyway it mostly downst supply enough power especially with ESP32 Boards.

closed 2021-03-28 by shyney7

duchengyao, 2023-05-23:
The same error in v60. I have tried short thick wires and use a power supply.

V58 work fine.