jameszah/ESP32-CAM-Video-Recorder-junior #10: I have an Error while compiling video recorder junior

Ak1167, 2021-08-18:
Hi guys.
I download esp32-cam video recorder junior code and going to upload it on module .
a few day ago I did it but now while compiling I have an error !
this is my errors :
" Arduino: 1.8.15 (Windows 7), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

In file included from E:\ESP32-CAM-Video-Recorder-junior-master\ESP32-CAM-Video-Recorder-junior-master\ESP32-CAM-Video-Recorder-junior-50x-lpmod\ESP32-CAM-Video-Recorder-junior-50x-lpmod.ino:1244:0:

C:\Users\elc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiManager.h:100:24: fatal error: strings_en.h: No such file or directory"

There is WifiManager.h and WifiManager.cpp codes in located address . What's reason of this error ??

[image]

jameszah, 2021-08-18:
Do you have WiFiManager installed as a library?   This is what I have:

[image]

Ak1167, 2021-08-18:
I have installed this libraray ....

[image]

thanks a million times.

jameszah, 2021-08-18:
There seem to be a lot of those - I used this one:

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ac3f0e212c76be264467904c7df336f7d9a3f71c/ESP32-CAM-Video-Recorder-junior-50x-lpmod.ino#L50

https://github.com/tzapu/WiFiManager

[image]

Ak1167, 2021-08-19:
Hi again.
I have another question that I could not answer it .
please look this serial port output : 
"
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.6
*WM: [3] hostname: STA:  desklen
Succesful Connection using WiFiManager

Internet status: 3
oLocal time: Thu Aug 19 14:58:14 2021
IP: 192.168.1.6
 
*WM: [3] unloading 
The power save was: 1
Set power save to 0
The power save is : 0
Setting up the camera ...
Frame config 13, quality config 5, buffers config 3
Before camera config ...Internal Total heap 301704, internal Free Heap 194292, SPIRam Total heap   4194252, SPIRam Free Heap   4194252
After  camera config ...Internal Total heap 301136, internal Free Heap 138024, SPIRam Total heap   4194204, SPIRam Free Heap   1314204

Camera started correctly, Type is 26 (hex) of 9650, 7725, 2640, 3660, 5640

Pic, len=77665, new fb 3F9D4C44
Pic, len=76549, new fb 3F8EA634
Pic, len=76184, new fb 3F800024
Pic, len=74533, new fb 3F9D4C44
Checking SD for available space ...
Total space: 7560MB
Used space: 40MB
Nothing deleted, 0.5% disk full
Starting Web Services ...
http task prio: 5
Camera http started
Creating the_camera_loop_task
the loop, core 0, priority = 3
the_sd_loop, core 1, priority = 4
  End of setup()

Shutting off wifi ...
*WM: [1] Disconnecting 
*WM: [3] WIFI station disconnect 
*WM: [3] unloading 
"

Why it disconnected from internet ??
I enter SSID and Pass in lines of code and compile it .
[image]

I can’t thank you enough.

jameszah, 2021-08-19:
Do you use the config.txt file on the SD card?  If it exists it replaces your compile-time parameters.
The default config.txt uses internet config "3" which gets the time, and then shuts off the internet.  Change that to 4, to keep the internet on ... unless you have a switch on pin 13 to control internet on/off.

You can modify the code below for your particular use.

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ac3f0e212c76be264467904c7df336f7d9a3f71c/ESP32-CAM-Video-Recorder-junior-50x-lpmod.ino#L2050

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ac3f0e212c76be264467904c7df336f7d9a3f71c/config.txt#L11

Ak1167, 2021-08-19:
Thanks for your attention.
I have seen this message on serial port : 
"Try to get parameters from config.txt ...
Failed to open config_file for reading
"
and check SD card , so there was no file config.txt in SD card.
next I follow code lines and read this line : 

"  int read13 = digitalRead(13);
  delay(20);
  read13 = read13 + digitalRead(13);  // get 2 opinions to help poor soldering"

then I connect IO13 to +3.3v

but nothing changed !

finally I Clear the issue and comment this lines :smile:  
[image]

It works but there is question that when I select Stream video speed is very slow but on recording on SD there isn't.
Is it because of connecting to WiFi ? 
so actually I want to use this module as a video recording device and some times connect and check it .
Is it possible to this module to check recorded videos without unmount SD card ?(May you put it somewhere is not accessible).

I apologize if I am not good in English .

Ak1167, 2021-08-19:
Is it necessary to put config.text file to SD card ??

jameszah, 2021-08-20:
You would ground pin 13 (through a 10k resistor) to enable/disable internet -- it is pulled high at the beginning.

If you are happy with your recording parameters, you can hard code them into the program, and you do not need a config.txt file on the SD.
If you are using your camera for various purposes (realtime/timelapse, HD/VGA/UXGA, ...), then you can set those parameters in the config.txt file on the SD, so you can modify the paramters with a text editor on your computer, and you do not need to reprogram the esp32.

The hardcoded stream delay in my program is here:

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ac3f0e212c76be264467904c7df336f7d9a3f71c/ESP32-CAM-Video-Recorder-junior-50x-lpmod.ino#L93

My sample config.txt stream delay is here:

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ac3f0e212c76be264467904c7df336f7d9a3f71c/config.txt#L10

If you are still using my compile time parameters, you are only getting 2 frames per second.
Switch is to "0" to get the frames as fast as the camera, and your wifi performance, will allow.

You might??? over-burden or over-heat the esp32 if you have the recording and streaming at maximum speed.  
Although I have tried to make them compatible, with a precedence going to the recording.

Ak1167, 2021-08-22:
Thanks a Million times :1st_place_medal: 
I am very excited that you answering all my questions with care and humility.
I do instructions you pointed and hope to do it right .

closed 2021-08-25 by Ak1167