jameszah/ESP32-CAM-Video-Recorder-junior #11: Is it possible to add FTP to Video-Recorder-junior version ??

Ak1167, 2021-08-24:
Hi Guys.
I finally could upload and run this module with this version (junior) .
I really appreciate your job.and now I have a problem.
How i can play recorded videos remotely ? 
In video-recorder-master version there is FTP ability to connect to module by FTP supporting software.( I tried to connect with ES Explorer on my Android gadget . but I wasn't success)
there is FTP server codes in recorder-master folder but I don't know how to add it in junior version.
I hope for your helps.
thanks all.

jameszah, 2021-08-24:
There is a version here with ftp:

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/issues/1#issue-740004083

The main changes are to start the ftp in the setup(), and call the handler once (plus or minus) per loop.
```

  ftpSrv.begin("esp", "esp");

  ftpSrv.handleFTP();
```

I found the ftp was quite slow for these massive avi files so did not add it here -- my use case for this program is battery powered, and mobile, so the long power hungry ftp transfer didn't suit it.

I have another program that will record short videos and send them to telegram, so there are accessible on your phone or computer, anywhere on the web.

https://github.com/jameszah/ESP32-CAM-Video-Telegram

Ak1167, 2021-08-25:
Thanks a lot Jameszah :+1: 
It is really good idea to send videos on telegram and see them every where you are.
But problem is here that 1.Telegram is forbidden in our country and we had to use proxy. 2.I don't know how to set proxy on my modem to make connecting device to telegram. do you know another application ?
And is it possible to play previous recorded video on telegram ?

Ak1167, 2021-08-28:
Thanks a lot 
I tried to send the photo taken with the camera via email with the module.it was successful.
Now I want to combine Junior with Email sender.
Because of I'm not using ESP32-cam-video recorder is that following picture ....

Low speed video recording and low quality .but junior version has a much better result.

[image]

Ak1167, 2021-09-02:
Thanks a lot.
I try it soon.

jameszah, 2022-01-13:
... new version better than ftp!

closed 2022-01-13 by jameszah