jameszah/ESP32-CAM-Video-Recorder-junior #12: Issues Compiling - conflicting declaration 'typedef enum HTTPMethod HTTPMethod'

AnthonyKNorman, 2022-01-22:
I get the following errors on compiling.
Please can you help?

```
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

ESP32-CAM-Video-Recorder-junior-58x.8:1452:3: error: conflicting declaration 'typedef enum HTTPMethod HTTPMethod'
 } HTTPMethod;
   ^
In file included from C:\Users\antho\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer\src/WebServer.h:30:0,
                 from G:\My Drive\Arduino\libraries\WiFiManager/WiFiManager.h:88,
                 from G:\My Drive\Arduino\ESP32-CAM-Video-Recorder-junior-58x.8\ESP32-CAM-Video-Recorder-junior-58x.8.ino:2:
C:\Users\antho\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer\src/HTTP_Method.h:6:26: note: previous declaration as 'typedef enum http_method HTTPMethod'
 typedef enum http_method HTTPMethod;
                          ^
G:\My Drive\Arduino\ESP32-CAM-Video-Recorder-junior-58x.8\ESP32-CAM-Video-Recorder-junior-58x.8.ino: In lambda function:
ESP32-CAM-Video-Recorder-junior-58x.8:1521:16: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
       if (info.wifi_sta_disconnected.reason != 201) {
                ^
ESP32-CAM-Video-Recorder-junior-58x.8:1522:98: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
         Serial.printf( "\nframe_cnt: %8d, WiFi event Reason: %d , Status: %d\n", frame_cnt, info.wifi_sta_disconnected.reason, WiFi.status());
                                                                                                  ^
ESP32-CAM-Video-Recorder-junior-58x.8:1523:98: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
         logfile.printf("\nframe_cnt: %8d, WiFi event Reason: %d , Status: %d\n", frame_cnt, info.wifi_sta_disconnected.reason, WiFi.status());
                                                                                                  ^
G:\My Drive\Arduino\ESP32-CAM-Video-Recorder-junior-58x.8\ESP32-CAM-Video-Recorder-junior-58x.8.ino: In lambda function:
ESP32-CAM-Video-Recorder-junior-58x.8:1546:98: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
         Serial.printf( "\nframe_cnt: %8d, WiFi event Reason: %d , Status: %d\n", frame_cnt, info.wifi_sta_disconnected.reason, WiFi.status());
                                                                                                  ^
ESP32-CAM-Video-Recorder-junior-58x.8:1547:98: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
         logfile.printf("\nframe_cnt: %8d, WiFi event Reason: %d , Status: %d\n", frame_cnt, info.wifi_sta_disconnected.reason, WiFi.status());
                                                                                                  ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\antho\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
conflicting declaration 'typedef enum HTTPMethod HTTPMethod'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

jameszah, 2022-01-22:
Sorry - there are 2 "ino" files in that folder - you have to delete the old one 58x.ino and keep 58x.8.ino -- I'll fix that shortly.

Also ver 58 is for esp32-arduino board library 2.0.2, but you are using 1.06, so switch to ver 57, or upgrade board library to 2.02 in Arduino - Tools - Boards - Board Manager - search for esp32 

[image]

jameszah, 2022-01-23:
deleted 58.8 -- it is now called 58

AnthonyKNorman, 2022-01-23:
Thank you.
Upgraded board to 2.0.2.
Used the correct ino.
Now compiles without errors.

closed 2022-01-24 by jameszah