jameszah/ESP32-CAM-Video-Recorder-junior #15: Can't find WifiManager.h

NicoPy, 2022-03-04:
Hi,

I'm trying to build the project.

First, it is not clear to me which .ino file to use. Is it the one at top level ? Or one in one of the sub-directories (v58, v57...) ?
I chose the top level one.

When clicking on "Verify", I get the following error :
```
Alternatives for WiFi.h: [WiFi@2.0.0]
ResolveLibrary(WiFi.h)
  -> candidates: [WiFi@2.0.0]
"C:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4-beta1-189-ga79dc75f0a\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/config" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/freertos/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/heap/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/log/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/lwip/include/apps" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/soc/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/soc/esp32" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/soc/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/hal/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/hal/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_rom/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_common/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_system/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/xtensa/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/xtensa/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/driver/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/driver/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_pm/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_ringbuf/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/efuse/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/efuse/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/vfs/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_wifi/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_event/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_netif/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_eth/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/tcpip_adapter/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_phy/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_phy/esp32/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_ipc/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/app_trace/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/esp_timer/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/mbedtls/port/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/mbedtls/mbedtls/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/mbedtls/esp_crt_bundle/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/app_update/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/spi_flash/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/bootloader_support/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/include/nvs_flash/include" "-IC:\\Users\\Nicolas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32/inc [... cut ...]
Using library EEPROM at version 2.0.0 in folder: C:\Users\Nicolas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\EEPROM 
Using library WiFi at version 2.0.0 in folder: C:\Users\Nicolas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi 
Compilation error: exit status 1
```
I tried to install libraries with WifiManager in the name (there are a lot) without success.
Which library should I use to get rid of this error ?

Notes :
- I'm using Arduino IDE 2.0.0-rc4
- I'm using ESP32 v2.0.2

jameszah, 2022-03-04:
Hmmm ... Arduino 2.0 -- I have not tried that one. -- I'm still on 1.8.19

You need the WiFiManager 2.05-beat by tablatronix installed through Arduino Tools Manage Libraries, search for WiFiManager and them slide down to the bottom and get this one (They are up to 2.0.9 now but I was using 2.0.5)

[image]

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/b1169bb7aea5afb721b0807aad8bf33840660215/v58/ESP32-CAM-Video-Recorder-junior-58x.ino#L52

You also need this one ArduinoJson 6.18.5 by Benoit Blanchon

You could also try the pre-compiled version.
https://jameszah.github.io/ESP32-CAM-VideoCam/

jameszah, 2022-03-04:
Also I see you are using the old 50x version -- it worked with old libraries, compilers.

If you have the new libraries compilers, you need the entire v58 folder, but nothing else.  Bring in the ,ino plus the cpp and .h files to the Arduino and it should have all the files it needs

NicoPy, 2022-03-04:
> Hmmm ... Arduino 2.0 -- I have not tried that one. -- I'm still on 1.8.19
> 
I'm just starting with Arduino and I thought it would be good to use up to date tools.

> You need the WiFiManager 2.05-beat by tablatronix installed through Arduino Tools Manage Libraries, search for WiFiManager and them slide down to the bottom and get this one (They are up to 2.0.9 now but I was using 2.0.5)
> 
This one is not in the list. I used the one by tzapu. It has compiled without error (not tested yet)

> [image]
> 
[image]

> https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/b1169bb7aea5afb721b0807aad8bf33840660215/v58/ESP32-CAM-Video-Recorder-junior-58x.ino#L52
> 
> You also need this one ArduinoJson 6.18.5 by Benoit Blanchon
> 
I have not installed it but it compiled without error.

> You could also try the pre-compiled version. https://jameszah.github.io/ESP32-CAM-VideoCam/
I'm quite sure I have clicked on the link in the project page but it did not redirect to the correct location. 
Now, it works. That's strange... I must be wrong.

NicoPy, 2022-03-04:
> Also I see you are using the old 50x version -- it worked with old libraries, compilers.
> 
> If you have the new libraries compilers, you need the entire v58 folder, but nothing else. Bring in the ,ino plus the cpp and .h files to the Arduino and it should have all the files it needs

Ok, That answers my first question. So I have to use the .ino in v58 folder.

NicoPy, 2022-03-04:
>>    You also need this one ArduinoJson 6.18.5 by Benoit Blanchon
> I have not installed it but it compiled without error.

Just found that this one is for ESP32-CAM-Video-Recorder (the non junior version).

NicoPy, 2022-03-04:
After a few tests it seems to be working.
Thanks for your support.

closed 2022-03-04 by NicoPy