jameszah/ESP32-CAM-Video-Recorder-junior #16: rotate the image on the camera

tiimsvk, 2022-03-10:
It is possible to implement flip or rotate image video settings? Thanks

jameszah, 2022-03-10:
Here is some commented-out code to horizontal and vertical flip the image.

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/b1169bb7aea5afb721b0807aad8bf33840660215/v58/ESP32-CAM-Video-Recorder-junior-58x.ino#L446

... and the full selection of camera configuration parameters .... a few lines above these are the values to put in the parameters.

https://github.com/espressif/esp32-camera/blob/86a4951f507e86034725b04f2428c98c49496dfe/driver/include/sensor.h#L203

Rotate is a little more difficult ... look here to see the different framesizes ... you can set it for hd (1280x720) or portrait-hd (720x1280) ... but I think the camera decides where the top of the frame is, and there is no hardware or driver software to turn it 90 degrees. 

https://github.com/espressif/esp32-camera/blob/86a4951f507e86034725b04f2428c98c49496dfe/driver/include/sensor.h#L91

giter404, 2022-07-07:
hey , i f u are using VLC player to view the video.avi file , u can turn the video by 180 degrees .
refer to this link.--
https://videoconverter.wondershare.com/vlc/how-to-rotate-a-video-using-vlc.html?gclid=CjwKCAjwiJqWBhBdEiwAtESPaJU4gs54P8qZ5CKcYWCicO40pnfUDfU6qoBA9UvALp5EZ022opYJqxoCp7oQAvD_BwE

closed 2025-02-26 by jameszah