jameszah/ESP32-CAM-Video-Recorder-junior #17: Text data overlay on video

andremgodoy, 2022-05-04:
Hello. Thanks for this amazing contribution !!
I'm currently building up a car camera with V58 and would like also to include some gps data in the bottom of screen like speed, direction angle and etc.. 
Where and how could I do this to the frame buffer?
Thank you in advance.
Andre M Godoy.

jameszah, 2022-05-05:
I'm adding that in a future release.  Basically, you have to convert every jpg from the camera to a bmp, then use some espressif routines to add the text, then convert back to jpg, and put it back into the stream that writes to the sd card.  It is quite time consuming and requires lots of psram for the buffers.  You might do every 10th or 100th frame, or put a small frame between 2 regular frames with just the text data.  Some video players accept that but others get confused.  I haven't really come up with a solution yet.  Stay tuned. ...

andremgodoy, 2022-05-06:
Indeed. I tried some combinations and even at CIF, I got around 1fps... not really feasible. Then, I have considered to perhaps create an additional text file with .srt extension containing the information in a format of subtitles.... I will try it out to see if it works.
Thanks again !

jameszah, 2022-05-06:
Good idea - I hadn't read the srt spec - it looks straightforward.  
Amusing that there seems to be nothing more official than this: 
http://forum.doom9.org/archive/index.php/t-73953.html

andremgodoy, 2022-05-11:
I tried some coding to create an SRT file and it's working like a charm =)
Thank you for your prompt answers !!

jameszah, 2022-09-15:
Some example code over here, if anyone is looking - it overlays time, altitude, acceleration for small rockets,

https://github.com/jameszah/ESP32-CAM-RocketCam

closed 2022-09-15 by jameszah