jameszah/ESP32-CAM-Video-Recorder-junior #18: Compiling error

navdeepssidhu, 2022-07-05:
First of all, thanks for great code. I am getting error while compiling ESP32-CAM-Video-Recorder-junior-50x-lpmod.ino ( error:  'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'? ) also when i run V58 sketch, its compiling fine and running well, but i am not able to change FPS and Length of video, by default cam is recording at fastest FPS and video length of 15 seconds. I tried to change the  values in sketch and config.txt as well. 
Can you please help.

jameszah, 2022-07-05:
The arduino_event disconnected thing looks like you need to upgarde your arduino board library to esp32-arduino 2.0.2 (or more)

The defualts in code are not used -- you have to set it in config.txt.  If you delete the config.txt it will create a new one with those default parameters.  Try editing the config.txt with the sd card in a computer, and then look at the arduino serial output to see if your new parameters are echoed after the reboot.  Or paste in your file below - it might be parsing it wrong?

navdeepssidhu, 2022-07-07:
I checked the board, it is esp32-arduino 2.0.3

dizcza, 2023-02-27:
I also have this issue. Tried esp32-arduino 2.0.5 & 2.0.7, same story.

Tried v60 but got
```
crc32.h: No such file or directory
```

Found CRC32.h in v58 but got
```
crc32.h: No such file or directory
```
Please mind the case: on Linux systems CRC32.h != crc32.h.

After I changed the include to "CRC32.h", I was able to compile and run v58.

One question. If I use your one-click-solution and don't specify Wifi creds (remove ssid and password lines), the Wifi won't launch (won't even attempt to launch), right? This is important as I actually don't want any wifi features because the RF waves screw up my compass recordings.

jameszah, 2023-02-27:
https://github.com/jameszah/ESP32-CAM-VideoCam/blob/62745b852d0812d3961a457de65939a1dfc67922/config.txt#L43

The no-wifi is an option here -- the one-click installer should create a config.txt as below, and the default wifi is ssid1234 which is no internet.

Looks like you have to edit the config.txt, as the default starts the hostmode (cinternet = 5)

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/976b276c28f2d28ec8ba44c9c43fa663b712f1c0/v60/ESP32-CAM-Video-Recorder-junior-60x.4.7/ESP32-CAM-Video-Recorder-junior-60x.4.7.ino#L653

I didn't know about the linux capitalization issue -- i'll correct that.

closed 2025-02-26 by jameszah