jameszah/ESP32-CAM-Video-Recorder-junior #19: issue basic

giter404, 2022-07-05:
hi sir , 
i downloaded zip of your code ,and opened v58 folder , then opened 'ESP32-CAM-Video-Recorder-junior-58x.ino'.
then while uploading code error showed . 
please help

jameszah, 2022-07-05:
I need a bit more information - like the arduino errors

jameszah, 2022-07-07:
https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/b1169bb7aea5afb721b0807aad8bf33840660215/v58/ESP32-CAM-Video-Recorder-junior-58x.ino#L52

[image]

giter404, 2022-07-07:
yo , it works now , thanks man i appreciate that .👍😊

giter404, 2022-07-07:
thanks man it works now 👍😊

closed 2022-07-07 by giter404

esalefu, 2022-07-27:
>
>
>
>
> thanks man it works now 👍😊

hi  what program work?