jameszah/ESP32-CAM-Video-Recorder-junior #21: downloading another file not working.

monroed11, 2022-09-13:
Hi, 
So I just downloaded 1 .avi file from http://IP:8080 but when I refreshed the page to get new file list, it is not responding anymore. http://IP/stream works fine, also http://IP. any input would be appreciated.

jameszah, 2022-09-14:
I have had some trouble downloading to chrome on android, but chrome on Windows 10 should work okay.  Another browser might work better.  Let me know your environment and what you see on the serial monitor during the problems.

It can also take quite a while if there are many files - try a clean sd, or drop everything into a sub-folder to the root is clean.

Here is the latest ... which should match the pre-compiled version.  Make sure you have esp32-arduino 2.04.

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/tree/master/v60

monroed11, 2022-09-14:
I am using Win10 and Chrome browser. for the serial monitor, no error/ related logs are displayed. I isolated the issue with this code: 
```
if (fileManager->streamFile(f, "application/octet-stream") != f.size()) {
          Serial.println(F("Sent less data than expected!"));
        }
```

It seams that this line does not stop. Tests I did was clean SD with just config, downloaded the config then refreshed the http://IP:8080 page. Its not loading anymore every after download. Thank you for providing the v60

monroed11, 2022-09-15:
I just tried your suggestion of using ESP32 V2.04 and it worked. Thank you very much!

closed 2022-09-15 by monroed11