jameszah/ESP32-CAM-Video-Recorder-junior #22: default - 1fps

punkapple123, 2022-09-16:
download source - upload default to esp32 (change ssid and password) - start video record - 1fps (micro sd(32GB and 1GB)) ;

jameszah, 2022-09-16:
Do you have this in your config.txt file?

The 0 interval should be record as fast as possible - which should be about 12 fps 720p on an ov2640 camera.

Maybe a slower camera? ov7670?

```
    desklens  // camera name
    11  // framesize  11=hd
    1800  // length of video in seconds
    0  // interval - ms between recording frames
    1  // speedup - multiply framerate
    0  // streamdelay - ms between streaming frames
    GMT // timezone
    ssid1234  // ssid wifi name
    mrpeanut  // ssid password
```

punkapple123, 2022-09-17:
yes, i have this in config.txt

Himrese, 2023-05-26:
got the same problem, is it solvable?

jameszah, 2023-05-26:
Can you post the serial monitor output, or the .txt file from the sd card with debug info.

jameszah, 2023-05-27:
Another thought - if the stream does not end normally it cannot accurately set the speed, so make the video length short enough that it ends, or end the video with the web page or the gpio

SaitamaTechno, 2024-08-19:
Hey buddy, firstly thank you for this great project. You should definitely put config.txt file inside v60 folder. I spent a few hours for searching updated config.txt file, finally here I found it on this issue. 
Secondly, make it use esp_cpu.h instead of soc/cpu.h. 
Thirdly, please write somewhere in the main readme file to change 3 lines at the code for 2mb psram support:
<br>
line numbers: 2482 2483 2484
```bash
framebuffer = (uint8_t*)ps_malloc(512 * 192); // buffer to store a jpg in motion // needs to be larger for big frames from ov5640
framebuffer2 = (uint8_t*)ps_malloc(512 * 192); // buffer to store a jpg in motion // needs to be larger for big frames from ov5640
framebuffer3 = (uint8_t*)ps_malloc(512 * 192); // buffer to store a jpg in motion // needs to be larger for big frames from ov5640

```
I searched through 3 issue pages to be able to run the arduino code. These updates will be helpful to a newbie.

And I have a special request: let us make timelapse over taking and saving images only, with an interval. This is actually easy, I already made one. And from my calculations, your ready speedup uses 2000mb  but my special request here will be using 150mb only, for 1 hour record.

jameszah, 2024-08-20:
Thanks for the polite comments.

1.  I tried to make it simple, so you would not need to copy the config.txt file onto the sd using a computer or edit it using a computer.  So the program creates the file if it does not exist, with all the default settings, and then you can edit it using the filemanager over wifi.  I can add a line of documentation about that.
2. 👍 
3. Bit of a sore area for all the people shouting at me to rewrite the program for their hardware.  I have a new version that guesses the size of the framebuffer, and generally improves memory allocation.  Haven't posted it yet.
4. Do you mean save the "jpg" on the sd card and you assemble them later?  

It is quite time-consuming to open and close files, and even copying thousands of files onto the computer at the end is very time consuming rather than the 1 avi file.

The avi doesn't require much space, just the original jpg's plus 8 bytes per frame for the index,

SaitamaTechno, 2024-08-20:
Thanks for response, about 4th step. Yes, I meant it. Because it really saves the space in the sd card and maybe it saves also battery capacity consuming a little power. I already wrote a program in python, jpeg to video converter. I tested with 3 hours recording, 1 image per second. it was 1280x720, 7000 photos. It didnt take so much time, maybe 2-3 minutes to copy and process the images into a video. But even without my timelapse request, your work is still very good, I love it. Thank you! Good Job!

closed 2025-02-26 by jameszah