jameszah/ESP32-CAM-Video-Recorder-junior #23: wifi setting

vipelz, 2022-09-17:
hello first of all thank you for this amazing project !!
after a few hours I recorded some videos !!
and I'm a real newbie ...

I would like to take a long time lapse video (about 6 months) and replace the battery every few days (7 or more is better).
but i don't know how to connect with wifi ..
I'm trying with version 60:
with my windows pc i can see the wifi "_desklens_c4_" appears but i can't connect to it.
- if i change in config.txt the ssid name should change the wifi name?

- maybe i should use this config with version 60?
https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/master/v58/config.txt

- also how can i have the wifi always on? Sorry but I do not understand
among this option which one is good for me:
_"3  // 0 no internet, 1 get time then shutoff, 2 streaming, using wifiman, 3 for use names below default off, 4 names below default on"_

sorry for all this question but I'm a really newbie guy.
again thank you very much sir for this great project !!

jameszah, 2022-09-17:
You can read more about version 60 over at 

https://github.com/jameszah/ESP32-CAM-VideoCam

There you can install the pre-compiled version and not worry about the code.  There are "lengthy install and operations instructions"

You can start it with an empty sd card, then it will write a config.txt, and then you can edit it -- i think it is the same as 58.

That "desklens_c4" name is in AP mode (no router involved - the esp32 doesn't connect to the internet - just your phone/computer) - the password is a hash of the mac address and shows on your serial monitor -- should look something like this:

https://github.com/jameszah/ESP32-CAM-RocketCam/blob/main/appass.png

Edit in you actual ssid and password, and then it will work through your router -- no more desklens_c4.

That "3//0 no internet" is old code to do a more general version, rather than the specialized version for that application.  Just use the config.txt file to set an ssid/password for station mode, or an automatic ssid (desklens_c4)  and hash password (2-gcf.....) for ap mode.

Regarding the 6 months thing - you should close the video - which writes the index - every 8 or 24 hours or so, and paste the little videos together at the end.  If you lose power, the video will be there on the sd, but there will be no index and it will be hard to deal with in your video editor.  And then when the esp32 reboots it will start a new video.

Many versions ... little documentation.

vipelz, 2022-09-17:
I just reset all and reinstall the pre-compiled so I was able to start wifiman and connect it to internet.
Now I'd like to put it in access point mode.
how can I put it in access point mode?

you are really very kind. 
Thank you

vipelz, 2022-09-17:
Ok I found it here is my Config.txt :

```

desklens  // camera name
11  // framesize  11=hd
21600  // ogni ora fa un videolength of video in seconds
5000  // interval - ms between recording frames 
1  // speedup - multiply framerate 
0  // streamdelay - ms between streaming frames
GMT // timezone
`**ap  // ssid wifi name**`
mrpeanut  // ssid password
```

thanks a lot man!!!

closed 2025-02-26 by jameszah