jameszah/ESP32-CAM-Video-Recorder-junior #26: Memory issue when run the code

Pantelis4722, 2022-10-07:
Hello,

Thanks for the project.
I am facing the following issues when i run the code : 

[E][camera.c:267] camera_fb_init(): Allocating 937 KB frame buffer Failed
[E][camera.c:1281] camera_init(): Failed to allocate frame buffer
E (18913) gpio: gpio_isr_handler_remove(398): GPIO isr service is not installed, call gpio_install_isr_service() first
[E][camera.c:1406] esp_camera_init(): Camera init failed with error 0x101

Any ideas what's the problem?
I use Arduino 1.8.18 Esp32cam

Thanks

jameszah, 2022-10-17:
Do you have the AI-Thinker chip with 4MB of PSRAM?
It should write out the psram on the serial monitor.

Pantelis4722, 2022-10-19:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
                                    
-------------------------------------
ESP-CAM Video Recorder v39

 http://desklens.local - to access the camera

-------------------------------------
Begin setup Core: 1
 Free Heap: 278384, priority = 1
 Himem is 2047KiB, Himem free 2047KiB, Flash is 4096KiB, Sketch is 996KiB 
 Write Q: 0 in/out  0 / 0

closed 2025-02-26 by jameszah