jameszah/ESP32-CAM-Video-Recorder-junior #29: Not getting stream.. please assist..

Lalbabu2001, 2023-03-09:
First of all ,very much thankyou for this amazing project, I am using v58 and  everything works fine except one.  I am not getting stream , and the pics of 3 sec delay.  What would ve reason?  
 Will the recording be deleted after the card full? Or not?  Thank you again!

jameszah, 2023-03-09:
You are probably blocking your wifi antenna.  The antenna is that square zigzag trace at the other end from the sd.  You have to keep it away from wires, metal, circuit board, which will swallow your wifi signal.  The adapter board esp32-cam-mb with the usb immediately under the antenna is also a problem.  You rssi number will show if you have bad wifi.  -30 is good, -80 is bad.

Lalbabu2001, 2023-03-24:
Hey, actually it's something like there is limit on wifi network side, even
close to module I'm getting about 300 KBps speed.. while downloading files.
i tried everything but the stream function and images stream doesnt work .
browser says cannot reach.. rest all works fine.

jameszah, 2023-03-25:
So you can find the web page, and download files (300 kBps sounds correct), but the series of jpgs does not work?  Do you get the title and time, but the image is not there .... it shows as a broken link to the jpg??  What browser / operating system are you using?  You could grab the http code from that page and look at the link to the jpg image with the "http ip address" etc to see where it us trying to find the jpg.

closed 2025-02-26 by jameszah