jameszah/ESP32-CAM-Video-Recorder-junior #30: Empty .avi file

FrancescoMontedori, 2023-03-29:
Hi, thanks for sharing your project. Im trying to use your sketch on esp32 cam and I am able to flash the sketch correctly. Anyway when i press the reset button to start the video recording and I open the serial monitor, i receive this message:

[image]

The Cam saves EMPTY .avi file and reboot, repeating the operation in loop.

Any suggestion?

jameszah, 2023-03-29:
I need the next page of the serial output .... the loop() part where it opens and closes the files.

FrancescoMontedori, 2023-03-30:
Hi, unfortunately there are no other pages in the serial output. As I said, the board is rebooting in loop. I attach a longer output:

[image]

Thanks again!

ikadatym-sry, 2023-04-05:
I also have the same issue. And this is the sd card read/write result.
[image]

jameszah, 2023-04-05:
How about send this section as text.  
And specify if you are using 60.4.7 and if it is the one click Installer version.

[image]

ikadatym-sry, 2023-04-05:
(I'm using 60.4.7 version) Compiled with Arduino IDE and your one click installer, either are giving me the same output.
==================================================================
Recording 1800 seconds.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008b954  PS      : 0x00060c30  A0      : 0x800d69dc  A1      : 0x3ffb3960  
A2      : 0x00000000  A3      : 0x3f864d90  A4      : 0x00007218  A5      : 0x00000000  
A6      : 0xe0ffd8ff  A7      : 0x464a1000  A8      : 0x00000000  A9      : 0x3ffb3910  
A10     : 0x3ffc9fc0  A11     : 0x3ffbf5f8  A12     : 0x3ffc9fc0  A13     : 0x00060c23  
A14     : 0x00060c20  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008b950  LEND    : 0x4008b96c  LCOUNT  : 0x00000720  

Backtrace: 0x4008b951:0x3ffb3960 0x400d69d9:0x3ffb3970

jameszah, 2023-04-05:
Hmmm ... I'll have to think about that.

```
PC: 0x4008b954
EXCVADDR: 0x00000000

Decoding stack results
0x400d69d9: the_camera_loop(void*) at C:\ArduinoPortable\arduino-1.8.19\portable\sketchbook\ESP32-CAM-Video-Recorder-junior\ESP32-CAM-Video-Recorder-junior-60x.4.7/ESP32-CAM-Video-Recorder-junior-60x.4.7.ino line 2615
```

[image]

ikadatym-sry, 2023-04-05:
Take your time. If you want me to test anything on my esp, you can tell me.

rkazemi711, 2023-06-07:
I have same issue...

duchengyao, 2023-09-16:
the same issue

```
A10     : 0x3ffc9670  A11     : 0x3ffbf5f8  A12     : 0x3ffc9670  A13     : 0x00060623  
A14     : 0x00060620  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008b930  LEND    : 0x4008b94c  LCOUNT  : 0x0000089f  

Backtrace: 0x4008b931:0x3ffb38a0 0x400d6a45:0x3ffb38b0

ELF file SHA256: cbd63ce5d85c2fa1

Rebooting...
ets Jul 29 2019 12:21:46
```

closed 2025-02-26 by jameszah