jameszah/ESP32-CAM-Video-Recorder-junior #31: Make it work with ESP32S3 model

FrancescoMontedori, 2023-04-21:
Hi, and thanks for sharing your project. I was able to make this sketch work on ESP32 CAM and it worked like a charm. However i would like to try to use this sketch on the ESP32S3 CAM. 
Have you ever tryed to upload the code on the ESP32S3 CAM?
I just tryed to upload the same code i uploaded to the ESP32 CAM but, as i expected, i receive this error:

[image]

My guess is that the SD card PIN on ESP32S3 CAM are different from the pin used in the ESP32 CAM, do you think that this could be the problem? In that case, which are the PIN used for the SD card in your sketch? 

Sorry for the dumb question but I am quite new to C programming.

jameszah, 2023-04-22:
Do you have a link to that device?  Schematic?

FrancescoMontedori, 2023-04-22:
Hi, thanks.
I attach the board schematic:

[image]

johnmcginn1, 2023-08-25:
Hi,

I would also like to know if its possible to use the code on ESP32S3 CAM or any other ESP device? Should the code be change and which parts should be changed? Also new to programming.

Thank you for your project so far!

UserID161, 2024-04-12:
The error says that the Pin has not been initialized, at the beginning of the program, add
#define SD_MMC_CMD  38 //Please do not modify it.
#define SD_MMC_CLK  39 //Please do not modify it. 
#define SD_MMC_D0   40 //Please do not modify it.
To the function static esp_err_t init_sdcard() add a line
SD_MMC.setPins(SD_MMC_CLK, SD_MMC_CMD, SD_MMC_D0);
before opening the SD card

jameszah, 2024-04-12:
https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC#pin-assignments-for-esp32-s3

That does not happened automatically with the correct arduino board library???
Which board library are you using?