jameszah/ESP32-CAM-Video-Recorder-junior #32: What‘s the major difference between these two project?

duchengyao, 2023-05-23:
Thanks for sharing this project, what is the main difference between the following two projects?

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior

https://github.com/jameszah/ESP32-CAM-Video-Recorder

Thanks ^_^

jameszah, 2023-05-27:
"junior" is newer, simpler, quicker, plus fully configurable after installing a generic version from https://jameszah.github.io/ESP32-CAM-VideoCam/

"original" has a some old inefficient code, but it does have the feature of sending the first frame of a video to telegram.org and your phone to give you some notice of the video being recorded.

Simple recording, or a starting place to make your own project, use "junior".

duchengyao, 2023-06-05:
This project has been successfully used, thank you for your help!

closed 2023-06-05 by duchengyao