jameszah/ESP32-CAM-Video-Recorder-junior #33: Timezone

Birrel01, 2023-11-16:
I can't get the time of the files right. I set GMT+1 (Amsterdam), but get GMT. Further on, how do I start with new file numbers?

jameszah, 2023-11-16:
Is that time on webpage or time on the files?
There was a daylight savings time bug in an old esp32-arduino?  Are you using the newest?

Reseting the numbers requires a "erase flash" and reinstall software.

Birrel01, 2023-11-16:
Set GMT-1, webpage is now showing right time, but file time is 1 hour behind. What do you mean with old esp32-arduino?

Birrel01, 2023-11-16:
Checked the files direct on the pc, and then the time is ok, so the time in the flemanager shows the wrong time.

jameszah, 2023-11-29:
https://github.com/espressif/arduino-esp32/issues/6786

You need the current esp32-arduino board manager library

[image]

closed 2025-02-26 by jameszah