jameszah/ESP32-CAM-Video-Recorder-junior #35: cant mount large format sd card?

ecto1a2003, 2024-01-14:
Brand new install with 64gb microsd fails as below, works with 2gb fine

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0

---
                                    
-------------------------------------
ESP32-CAM-Video-Recorder-junior v58
-------------------------------------
setup, core 1, priority = 1
--- reboot ------ because: ESP_RST_POWERON
Reading the eprom  ...
Good settings in the EPROM 
New File Group 9
Writing to EPROM ...
Mounting the SD card ...
E (957) vfs_fat_sdmmc: mount_to_vfs failed (0xffffffff).
Failed to mount SD card VFAT filesystem. 
Do you have an SD Card installed?
Check pin 12 and 13, not grounded, or grounded with 10k resistors!

ecto1a2003, 2024-01-16:
Bummer, bo support for exfat? Ill see i i can even find smaller cards

jameszah, 2024-01-16:
The 64GB cards work fine, you just have to format it fat32.

[image]

ecto1a2003, 2024-01-16:
Hate to waste them, thanks.

closed 2025-02-26 by jameszah