jameszah/ESP32-CAM-Video-Recorder-junior #36: add ablility to control light?

ecto1a2003, 2024-01-15:
would it be possible to control the light on these?

leahcornelius, 2024-07-06:
The flash light on the front is connected to GPIO 4, if you use analogWrite you can dim it as it is VERY bright or if you want a SERIOUSLY bright light just set the pin mode in the setup() function and use digitalWrite(4, HIGH/LOW) to control it

jameszah, 2024-07-06:
Are you doing the Aerospace thing?
You might start with this that records temperature, altitude, and acceleration on top of the video:
https://github.com/jameszah/ESP32-CAM-RocketCam?tab=readme-ov-file#esp32-cam-rocketcam

leahcornelius, 2024-07-18:
@jameszah I am, thank you for pointing that repo out i will have a look at its code for pointers however my flight computer architecture is quite more complex and modular. The ESP-CAM is simply responsible for video recording (with some other fancy features i wont go into here, unless you are interested. It is refereed to as the videography-interface-module AKA VIM of which there maybe several for large HPR flights), turned on and off by the master flight controller (MFC). 

The MFC handles all the aspects of measuring sensors, deriving altitude, 3d orientation and comparing to the simulated flight plan to check for abnormalities (it is able to detect parachute deployment failures based on acceleration and velocity and follow backup procedures, prevent staging if over a set angle to prevent the second stage being fired if the rocket flips during flight, and interfaces with the telecom control system (TCS) which can transmit data over GSM, LoRa and a custom overlay on the ESP-MESH protocol.

So essentially VIM(s) -> MFC
Sensors -> MFC = calculated data on ascent path, velocity and altitude; This is ran through checks at critical stages like drouge deployment or staging to check for abnormalities as well as the uploaded flight simulation to check for significant deviations in performance of our motors.

MFC  (telemetry and location data) -> TCS 
TCS (ground station responses or abort commands) -> MFC 

So far more complex than it all running on one board as we do A-LOT of calculations to enable future development (eg thrust vector control, active guidance via fin control etc) and the MFC transmits data to the computer after launch on which the software stitches footage, data and visuals to produce overlays of data, multi camera view etc as desired.

Thank you for the link though, i will check it out as i am very much new to the AVI protocol. And thanks for the heads-up on GPIO4, we will make sure only to use it for on-ground indication when the SD is not being read. 

Kind regards,
Leah C on behalf of Cornelius Aerospace