jameszah/ESP32-CAM-Video-Recorder-junior #38: Just trying to contact jamesZah

Cyrille-de-Brebisson, 2024-02-13:
Hello,

Sorry for going this route, but I am trying to contact you and github does not have any email system anymore :-(

I saw that you were working on trying to get RAW data from various OV chips on an ESP32. I am trying to do the same thing with no success and I was wondering if you ever succeeded?

Thanks,
Cyrille.de.brebisson@gmail.com

jameszah, 2024-02-14:
Hi,
I looked through the code a little, but did not bypass the driver to fetch the data.

https://github.com/espressif/esp32-camera/blob/master/driver/sccb.c

You can set the registers in the camera ov2640 and control it somewhat .. like how fast the lines are read on the sensor and things, but I still use the driver to fetch the data.

There was some discussion about accessing the registers of the ov5640, but the internal documentation on that is even worse than available material on the ov2640.

What are you trying to access?

Cyrille-de-Brebisson, 2024-02-14:
Hello,

What I want to do is use either an OV2640 OR an OV5640 (in bin2*2 mode) to get RAW bayer data and send them through wifi.

It should, in theory, be simple. But lack of documentation on the chip and source code makes it impossible...

I seem to be able to get RAW data from the 5640, BUT the format is not what I am expecting, and I have not be able to decode it yet. AND the esp system crashes when I try to get higher res that 320*240...

Getting RAW from the 2640 is impossible as there is no documentation...

Cyrille

jameszah, 2024-02-14:
Hmmm ... beyond my experience.  
I assume you have this document:  https://www.uctronics.com/download/cam_module/OV2640DS.pdf

The "raw" is like an rgb888 with unprocessed colors???
The esp32-cam driver has some discussion about there being too much data to download large frames as rgb, so they recommend get the jpg and convert back.  The raw would be bigger than the rgb888 for a full sensor (1632x1232x 8bits?), so you might have to look at those sccb routines to  see why they cannot handle that data.

Cyrille-de-Brebisson, 2024-02-15:
Hello,

Yes, I have the pdf, but it is only partial. It misses a lot of register definition. And I was 100% unable to get the full version, even asking through my company which does business with OmniVision! Apparently, they stopped prducing the 2640 5+ years ago... So it makes you wonder where the stock that we buy on aliExpress comes from! I do not understand OmniVison maladive secrecy around their chips documentation...

Anyhow, back to the question. RAW is actually 1/3 of the size of RGB888.
The sensor array (1632*1232) has only ONE color per pixel (there is a color filter in front of each photosite).
When you get 1600*1200 RGB pixels, only ONE of thee 3 component is known for said pixel and the 2 others are interpolated from the nearby pixels with the correct color.
So, RAW is actually 3 times SMALLER than RGB888...

So a RAW frame would be just shy of 2MB. And would then need to be sent through wifi.
Now, I do NOT need speed, so there is no issue in having to take a frame, send it and start again... I do not need throughput...

I am trying to use a OV5640 right now as it looks like there is more doc on the chip... But the support in the espLib is still limited :-(

closed 2025-02-26 by jameszah