jameszah/ESP32-CAM-Video-Recorder-junior #43: Try on Freenove ESP32-S3 Wroom

sakana1110, 2024-11-01:
I am a beginner and try to modify the version you have released. I can successfully upload,but not working and the serial monitor shows
[image]
I modified "avi_length=10", camera pin,deleted major_fail(), deleted the part controlled by pin12 and set "stop=1" (this should automatically stop recording after 10 seconds), add "  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;".attached the modified file , thank you in advance for your answer.
```
//Change from https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/master/old/ESP32-CAM-Video-Recorder-junior-02x.ino
static const char vernum[] = "v02";
static const char devname[] = "desklens";         // name of your camera for mDNS, Router, and filenames

int  framesize = 7;                //  10 UXGA, 9 SXGA, 7 SVGA, 6 VGA, 5 CIF
int  quality = 20;                 //  quality on the 1..63 scale  - lower is better quality and bigger files - must be higher than the jpeg_quality in camera_config
int avi_length = 10;               // how long a movie in seconds

int MagicNumber = 11;                // change this number to reset the epron in your esp32

//#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
#include "esp_log.h"
#include "esp_http_server.h"
#include "esp_camera.h"

#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM     15
#define SIOD_GPIO_NUM     4
#define SIOC_GPIO_NUM     5
#define Y2_GPIO_NUM       11
#define Y3_GPIO_NUM       9
#define Y4_GPIO_NUM       8
#define Y5_GPIO_NUM       10
#define Y6_GPIO_NUM       12
#define Y7_GPIO_NUM       18
#define Y8_GPIO_NUM       17
#define Y9_GPIO_NUM       16
#define VSYNC_GPIO_NUM    6
#define HREF_GPIO_NUM     7
#define PCLK_GPIO_NUM     13

camera_fb_t * fb_curr = NULL;
camera_fb_t * fb_next = NULL;

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"
#include "soc/soc.h"
#include "esp_cpu.h"
#include "soc/rtc_cntl_reg.h"

static esp_err_t cam_err;

//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
//  Avi Writer Stuff here

// MicroSD
#include "driver/sdmmc_host.h"
#include "driver/sdmmc_defs.h"
#include "sdmmc_cmd.h"
#include "esp_vfs_fat.h"
#include <SD_MMC.h>

FILE *avifile = NULL;
FILE *idxfile = NULL;

long bp;
long ap;
long bw;
long aw;
[... 707 lines cut ...]

    float fps = frames / ((avi_end_time - avi_start_time) / 1000) ;
    Serial.printf("End the avi at %d.  It was %d frames, %d ms at %.1f fps...\n", millis(), frames, avi_end_time, avi_end_time - avi_start_time, fps);

    frames = 0;             // start recording again on the next loop

  } else {  // another frame of the avi

    fb_curr = fb_next;           // we will write a frame, and get the camera preparing a new one

    fb_next = get_good_jpeg();    // should take near zero, unless the sd is faster than the camera, when we will have to wait for the camera

    another_save_avi(fb_curr);

    esp_camera_fb_return(fb_curr);

    if (frames % 100 == 0 ) {     // print some status every 100 frames

      float fps =   1.0 * frames / ((millis() - avi_start_time) / 1000) ;
      Serial.printf("So far: %d frames, %d ms at %.2f fps...\n", frames, millis() - avi_start_time, fps);
    }
  }
}

```

jameszah, 2024-11-03:
If it gets stuck starting the camera, you probably have your gpio<->camera pins wrong.
Is there are simple test program for that board, just to show the jpegs to a web browser.
Have you also checked the sd gpio pins - they would have changed as well.

Is this what you used

https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board/blob/main/ESP32S3_Pinout.png

You could turn on the DEBUG messages in Arduino to get some more messages about what is failing ...

[image]

Which Arduino board type are you using?

You could also switch to the much newer code that I am more familiar with - like version 60 -- and just delete any sections that you dont need.

sakana1110, 2024-11-03:
Yes, this is what I used, but I refer to another example (https://github.com/Freenove/Freenove_Ultimate_Starter_Kit_for_ESP32_S3). I tried to use example 32.2 in it and it ran normally. Then I modified the pins of the SD card (not sure is the change work).Thank you for your response. With my current abilities, I'm unable to proceed further.
[image]
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
[   113][I][esp32-hal-psram.c:90] psramInit(): PSRAM enabled
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 2
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   393444 B ( 384.2 KB)
  Free Bytes        :   365384 B ( 356.8 KB)
  Allocated Bytes   :    23156 B (  22.6 KB)
  Minimum Free Bytes:   360336 B ( 351.9 KB)
  Largest Free Block:   327668 B ( 320.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8385672 B (8189.1 KB)
  Allocated Bytes   :      576 B (   0.6 KB)
  Minimum Free Bytes:  8385672 B (8189.1 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
-----------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  8388608 B (8 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Nov  3 2024 11:51:07
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.4-358-gbd2b9390ef-dirty
[... 795 lines cut ...]
    avi_end_time = millis();

    float fps = frames / ((avi_end_time - avi_start_time) / 1000) ;
    Serial.printf("End the avi at %d.  It was %d frames, %d ms at %.1f fps...\n", millis(), frames, avi_end_time, avi_end_time - avi_start_time, fps);

    frames = 0;             // start recording again on the next loop

  } else {  // another frame of the avi

    fb_curr = fb_next;           // we will write a frame, and get the camera preparing a new one

    fb_next = get_good_jpeg();    // should take near zero, unless the sd is faster than the camera, when we will have to wait for the camera

    another_save_avi(fb_curr);

    esp_camera_fb_return(fb_curr);

    if (frames % 100 == 0 ) {     // print some status every 100 frames

      float fps =   1.0 * frames / ((millis() - avi_start_time) / 1000) ;
      Serial.printf("So far: %d frames, %d ms at %.2f fps...\n", frames, millis() - avi_start_time, fps);
    }
  }
}
```

jameszah, 2024-11-03:
Try lowering fb count to 3 (maybe 4), and quality up to 12, and do a print statement after the camera_init line to see if that is where it is stuck. 

[image]

sakana1110, 2024-11-03:
Change like this?
[image]
[image]
[image]
It seems that there is a problem with the following parts:
[image]

jameszah, 2024-11-04:
You could change from size from UXGA to HD, just above the fb_count change, and include that serial print after the fb_get line ... that will tell you if the camera is working.  Otherwise you will have to pin down which of those 3-4 lines is causing the problem.

After that you will need to look at the init_sdcard code to see how much of it is running before it stops.  I see you are setting the gpios there.  Are you using a freenove-cam-s3 board definitiions?  I think they would be set there, but dont see any board definitions on my arduino ide.

I assume there is no more serial output after the "after camera init" line?  No garbage or anything printed?
The serial output might be disturbed.

sakana1110, 2024-11-04:
Thanks for your response. I have found something new,it is stuck here.
[image]
And after I comment out.It show this.
```
/*
for (int j = 0; j < 5; j++) {
    camera_fb_t * fb = esp_camera_fb_get();
    Serial.print("Pic, len="); Serial.println(fb->len);
    esp_camera_fb_return(fb);
    delay(50);
  }
  */
```

```
-------------------------------------
ESP32-CAM-Video-Recorder-junior v02
-------------------------------------
setup, core 1, priority = 1
Internal Total heap 393444, internal Free Heap 363584
SPIRam Total heap   8388608, SPIRam Free Heap   8385412
Setting up the camera ...
after camera initGuru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x420029b8: 8565f4a1 00000002 31004136
Core  1 register dump:
PC      : 0x420029bd  PS      : 0x00060030  A0      : 0x82006ea6  A1      : 0x3fcebf20  
A2      : 0x3fcec9c0  A3      : 0x420164a8  A4      : 0x00800000  A5      : 0x00000001  
A6      : 0xffffffff  A7      : 0x3fc93398  A8      : 0x820029bd  A9      : 0x3fcebf00  
A10     : 0x000001f4  A11     : 0x3c04c081  A12     : 0x0000000f  A13     : 0x3c04c081  
A14     : 0x00000000  A15     : 0x00000026  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420029ba:0x3fcebf20 0x42006ea3:0x3fcebfc0 0x4037df2e:0x3fcebfe0
```

sakana1110, 2024-11-04:
I add more print to know where is the problem.
```
static esp_err_t config_camera() {

  camera_config_t config;

  //Serial.println("config camera");

  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.grab_mode = CAMERA_GRAB_LATEST;

  config.xclk_freq_hz = 20000000;     // 10000000 or 20000000 -- 10 is faster !!

  config.pixel_format = PIXFORMAT_JPEG;

  config.frame_size = FRAMESIZE_HD; // edit in framesizes below -- this must be better than the framesize specified at the top
  /*
      FRAMESIZE_96X96,    // 96x96
      FRAMESIZE_QQVGA,    // 160x120
      FRAMESIZE_QCIF,     // 176x144
      FRAMESIZE_HQVGA,    // 240x176
      FRAMESIZE_240X240,  // 240x240
      FRAMESIZE_QVGA,     // 320x240
      FRAMESIZE_CIF,      // 400x296
      FRAMESIZE_HVGA,     // 480x320
      FRAMESIZE_VGA,      // 640x480
      FRAMESIZE_SVGA,     // 800x600
      FRAMESIZE_XGA,      // 1024x768
      FRAMESIZE_HD,       // 1280x720
      FRAMESIZE_SXGA,     // 1280x1024
      FRAMESIZE_UXGA,     // 1600x1200
  */

  config.jpeg_quality = 12;  // 1 to 63 - smaller number is higher quality and more data - must be lower rhat the quality parameter at the top
  
  config.fb_count = 2;

  // camera init
  cam_err = esp_camera_init(&config);
  if (cam_err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", cam_err);
  }
  Serial.printf("after camera init");
  

  sensor_t * ss = esp_camera_sensor_get();
  ss->set_quality(ss, quality);
  ss->set_framesize(ss, (framesize_t)framesize);

  ss->set_brightness(ss, 1);  //up the blightness just a bit
  ss->set_saturation(ss, 0); //lower the saturation

  delay(500);
  
  for (int z = 0; z < 5; z++) {
    camera_fb_t * fb = esp_camera_fb_get();
    Serial.print("Pic, len="); Serial.println(fb->len);
    esp_camera_fb_return(fb); 
    delay(50);
  }
  Serial.print("after return test finish");

}

```
```
void setup() {
  Serial.begin(115200);
  Serial.println("\n\n---");

  //Serial.setDebugOutput(true);

  // zzz
  Serial.println("                                    ");
  Serial.println("-------------------------------------");
  Serial.printf("ESP32-CAM-Video-Recorder-junior %s\n", vernum);
  Serial.println("-------------------------------------");

  Serial.print("setup, core ");  Serial.print(xPortGetCoreID());
  Serial.print(", priority = "); Serial.println(uxTaskPriorityGet(NULL));

  // put your setup code here, to run once:
  Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  Serial.printf("SPIRam Total heap   %d, SPIRam Free Heap   %d\n", ESP.getPsramSize(), ESP.getFreePsram());

  Serial.println("Setting up the camera ...");

  config_camera();
  Serial.println("after camera config");

  Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  Serial.printf("SPIRam Total heap   %d, SPIRam Free Heap   %d\n", ESP.getPsramSize(), ESP.getFreePsram());

  // SD camera init
  esp_err_t card_err = init_sdcard();
  if (card_err != ESP_OK) {
    Serial.printf("SD Card init failed with error 0x%x", card_err);
    return;
  }

  Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  Serial.printf("SPIRam Total heap   %d, SPIRam Free Heap   %d\n", ESP.getPsramSize(), ESP.getFreePsram());

  Serial.println("Warming up the camera ... here are some frames sizes ...");

  for (int i = 0; i < 10; i++) {
    camera_fb_t * fb = esp_camera_fb_get();
    Serial.printf("frame %d, len %d\n", i, fb->len);
    esp_camera_fb_return(fb);
    delay(100);
  }

  do_eprom_read();

  Serial.println("  End of setup()\n\n");

  boot_time = millis();
}
```
```
-------------------------------------
ESP32-CAM-Video-Recorder-junior v02
-------------------------------------
setup, core 1, priority = 1
Internal Total heap 393444, internal Free Heap 363584
SPIRam Total heap   8388608, SPIRam Free Heap   8385412
Setting up the camera ...
after camera initPic, len=5076
Pic, len=5076
Pic, len=5076
Pic, len=5074
Pic, len=5074
after return test finishGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420029fa  PS      : 0x00060030  A0      : 0x82006ee2  A1      : 0x3fcebf20  
A2      : 0x3fc97450  A3      : 0x00000000  A4      : 0x3c060d48  A5      : 0x00000001  
A6      : 0xffffffff  A7      : 0x3fc93398  A8      : 0x820029fa  A9      : 0x3fcebf00  
A10     : 0x00000018  A11     : 0x3c0402c4  A12     : 0x0000000a  A13     : 0x3c04c0a1  
A14     : 0x00000000  A15     : 0x00000026  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x82006ee0  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffff9  

Backtrace: 0x420029f7:0x3fcebf20 0x42006edf:0x3fcebfc0 0x4037df2e:0x3fcebfe0
```

jameszah, 2024-11-04:
If that for loop does not work, that means the memory is scrambled up - something is dumping random information in ram.

You could use exception decoder to see what it is doing when your get the illegal instruction or load prohibited.

https://github.com/me-no-dev/EspExceptionDecoder

And the frames are too small and all the same size ... so camera is not working and memory is scrambled.

You might try taking your working example, and adding in some code to mount the sd card, and then add some code to write the avi, ... all on top of the working example, that does not have the scrambled memory.

sakana1110, 2024-11-05:
謝謝你的耐心和回答，我想轉用V60試看看，在我修改完相機、SD卡引腳，刪除多餘的引腳設定後，在"new_simple.print(config_txt);"報錯:"Compilation error: 'config_txt' was not declared in this scope"

jameszah, 2024-11-05:
You have to frag all the files into the arduino ide, like this:
[image]

sakana1110, 2024-11-07:
My bad, I missed one. After that I get an error when I run it and I can't find the cause of the problem:
[image]
Part of code:
```
void the_camera_loop (void* pvParameter) {

  Serial.print("the camera loop, core ");  Serial.print(xPortGetCoreID());
  Serial.print(", priority = "); Serial.println(uxTaskPriorityGet(NULL));

  frame_cnt = 0;
  start_record_2nd_opinion = 1;
  start_record_1st_opinion = 1;
  start_record = 0;

  delay(1000);

  while (1) {

    // if (frame_cnt == 0 && start_record == 0)  // do nothing
    // if (frame_cnt == 0 && start_record == 1)  // start a movie
    // if (frame_cnt > 0 && start_record == 0)   // stop the movie
    // if (frame_cnt > 0 && start_record != 0)   // another frame

    ///////////////////  NOTHING TO DO //////////////////
    if (frame_cnt == 0 && start_record == 0) {

      // Serial.println("Do nothing");
      if (we_are_already_stopped == 0) Serial.println("\n\nDisconnect Pin 12 from GND to start recording.\n\n");
      we_are_already_stopped = 1;
      delay(100);

      ///////////////////  START A MOVIE  //////////////////
    } else if (frame_cnt == 0 && start_record == 1) {

      //Serial.println("Ready to start");

      we_are_already_stopped = 0;

      //delete_old_stuff(); // move to loop

      avi_start_time = millis();
      Serial.printf("\nStart the avi ... at %d\n", avi_start_time);
      Serial.printf("Framesize %d, quality %d, length %d seconds\n\n", framesize, quality, avi_length);
      logfile.printf("\nStart the avi ... at %d\n", avi_start_time);
      logfile.printf("Framesize %d, quality %d, length %d seconds\n\n", framesize, quality, avi_length);
      logfile.flush();

      frame_cnt++;

      long wait_for_cam_start = millis();
      fb_curr = get_good_jpeg();                     // should take zero time
      wait_for_cam += millis() - wait_for_cam_start;

      start_avi();

      wait_for_cam_start = millis();
      fb_next = get_good_jpeg();                    // should take nearly zero time due to time spent writing header
      //if (framebuffer_time < (millis() - 10)){
      xSemaphoreTake( baton, portMAX_DELAY );
      framebuffer_len = fb_next->len;                    // v59.5
      memcpy(framebuffer, fb_next->buf, fb_next->len);   // v59.5
      framebuffer_time = millis();                       // v59.5
      xSemaphoreGive( baton );
      //}
      wait_for_cam += millis() - wait_for_cam_start;
      xSemaphoreGive( sd_go );                     // trigger sd write to write first frame

      ///////////////////  END THE MOVIE //////////////////
    } else if ( restart_now || reboot_now || (frame_cnt > 0 && start_record == 0) ||  millis() > (avi_start_time + avi_length * 1000)) { // end the avi

[... 163 lines cut ...]
      sprintf(localip, "%s", WiFi.localIP().toString().c_str());
      Serial.print("IP: "); Serial.println(localip); Serial.println(" ");
    }

    if (!MDNS.begin(devname)) {
      Serial.println("Error setting up MDNS responder!");
    } else {
      Serial.printf("mDNS responder started '%s'\n", devname);
    }
  }

  if (reboot_now == true) {
    Serial.println(" \n\n\n Rebooting ... \n\n\n");
    delay(2000);
    ESP.restart();
  }
  if (!InternetOff) {
    filemgr.handleClient();  //v56
  }
  delay(200);

}
```
By the way, I really appreciate your help, so I bought you a cup of coffee, please enjoy
[image]

jameszah, 2024-11-08:
I got one of those modules.  

The changes are the camera gpio, the sd gpio, the stack business needs some more bytes, enable the psram, and disable the brownout changes.

You seem to have solved it, so I wont post for now.

Regarding the record on event things.  You can check your readings every run through the loop, and set the start_record flag.  Or if your device has a interrupt you can control the start_record flag from the interrupt handler.

There is some code here:

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/b6c9af8970bc2321935662286603da7ce6484271/vA1/TimeLapseAviA1x.ino#L529

The loop is simpler and plenty fast, as the next photo is 80 ms at HD, so the very loop will run plenty of times every 80ms.

sakana1110, 2024-11-10:
All functions are working normally. Thank you again for your help. However, I want to increase the frame rate of recording, so I am interested in enabling the psram. Can you elaborate on it? Currently, my picture quality is set to HD, and I added "config.grab_mode = CAMERA_GRAB_LATEST;" "config.fb_location = CAMERA_FB_IN_PSRAM;" (but it seems there is no change). The number of frames recorded is about 10fps. I want to know if it can be increased to about 25.

jameszah, 2024-11-10:
Psram does not make it faster ... it is slower than regular ram, but not really relevant as the bottleneck is the camera or the sd card, or sometimes the esp32 processor speed.

To eanble the psram, set the setting below.

The camera can operate at 12.5 fps is HD, or 25 fps SVGA (800X600) or below, and 50 fps CIF (400x296) or below.
You could get an ov5640 that works with the esp32, which will give you better resolution and about twice the speed ... 24 fps HD, but you will run into the sd card bottleneck at some point, which may or may not improve with faster sd cards.

The sd card can usually keep up with the ov2640 rates, but some sd csrds are slower, or if the sd is fragmented, or other processing  on the esp32 (like streaming or downloading videos) can slow things down.

I fixed that little program you were using originally - I can post that if you like.  It still has some old code, so I was thinking of rewriting it to make it very simple. 

[image]

sakana1110, 2024-11-14:
Got it, thank you for your answer, I really appreciate your help, I will close this issues since there are no more questions.

closed 2024-11-14 by sakana1110