jameszah/ESP32-CAM-Video-Recorder-junior #44: Questions about AVI header compression field

try-except, 2024-12-11:
Hi! I'm developing a small video recorder with ESP32-CAM and I've been looking at the code in this repo (thank you very much for making this, it's been of great help), but I cant' figure out why you wrote "MJPG" in the Compression field of the strf list. I searched online but found that there's only a handful of valid values for that field [like the ones here](https://learn.microsoft.com/en-us/openspecs/windows_protocols/ms-wmf/4e588f70-bd92-4a6f-b77f-35d0feaf7a57).

Thanks again for the amazing code and sorry if this question is a bit dumb, I hadn't worked with the AVI format before.

jameszah, 2024-12-11:
Hmmm ... I do not know.  I copied that header from a arducam example after my first version which was a mjpeg recorder.  It worked, so I moved on.  Not great documentation on avi ... the more modern mp4 format is more complicated, but better documentation.

try-except, 2024-12-13:
I understand, no worries :)

askchamp, 2025-02-18:
i was successfully able to use the vids shot-on this cam in this vid - https://youtu.be/Z4Fiy3cJaNs?si=xrgji3H-6pi_rcgX

closed 2025-02-26 by jameszah