jameszah/ESP32-CAM-Video-Recorder-junior #46: Trigger and Stop recording

barbierbrice, 2025-02-18:
Hello there,

I have been successfully installing the setup and I have been able to stream and record videos with this git.
So first of all, thanks a lot for sharing this.

I have a couple of questions:

- would it be possible not to record all the time but add a trigger button on the web page ? I would like to trigger the video when set and stop it when not needed. I don't need it to permanently record.
- is it possible to change the recording codec? MJPG is not Mac friendly.
- what are your latency when streaming on browser? I would like the latency to be reduced to the minimum if possible.

So, thanks again
cheers

jameszah, 2025-02-26:
web stop/start is here - you could also have another device send the "192.168.1.100/start"

You can also add a hardware device to gpio 12 - when it is grounded the video will write index and stop, and then when it is released from ground, it will start recording again
 
https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/tree/248059f023b5a0c182fd39c73d512947f8c12512/v62

mp4 would be a nicer codec, but quite a bit a work 😄 

latency should be good - really depends on esp32<->router distance and walls etc.  It is quicker if you access the esp32 through AP mode where you do not go through the router.  Also smaller framesize will speed it up.

barbierbrice, 2025-02-26:
Thank you for your reply.
I have been working on theses points a bit further.

MJPEG allows me to keep the resolution quite high and will work well for the project I'm involved on.
I have found [this page ](https://www.linkedin.com/posts/pleasedontcode_exploring-h264-encoding-on-esp32-a-practical-activity-7251957212867170306-3hwQ/) which go through a method to get H264. 

Start and Stop URIs work perfectly well as I have realised after my previous post.
Will think about adding a hardware device though.

I have adapt this part to use a JS syntax instead but that's already working fine.
When coding the html part, the code was sometimes bugging the whole board for no reason.
Do you know if there is any limitation regarding the HTML implementation in the ESP32 ?

Unfortunately I have to use a router as I have multiple cameras running at the same time.
We will accept the latency.

Quick one : is there a way to set the framerate to 24fps?
I have been playing around with the config.txt but could not set a proper 24.000fps timebase.

I have as well tried the OV5640 to replace the standard OV2640.
I have mixed feeling about it : the resolution is increased but the quality of the stream seems to be quite bad.
What do you think about it?

In the CameraWebserver-ESP32 template, there are a couple of video adjustments which could be interesting added to your code. Would the board allows to get new settings?

Thanks again,

jameszah, 2025-02-27:
24fps 
2640 only does 12.5 fps at hd, so switch to vga and zero recording interval and you will get 24fps recording.  There is still the bottleneck of the sd card - cheap aliexpress cards usually work, but have a few slow spots.  The new version of the code keeps track of the slow spots.  Thinking of doing a formatter program to find and occupy the slow spots.

I'll have a look at h264, but I see they want you to use 320x240 frames, rather than 1280x720.  Converting the hardware manufactured jpeg to bmp is pretty quick, but software encoding back to a jpeg or other (whatever 264 is) is very slow -  maybe 1 or 2 fps.

jameszah, 2025-02-27:
Also streaming is dependent on wifi.  hd can is limited by camera at 12.5, but vga 24fps rarely gets above 18 in perfect-ish conditions.  Smaller frames could get to 24 easily.

jameszah, 2025-03-01:
That h264 stuff seems to be esp32p4 risc chip with hardware encoding -- it says 80fps 720p vs about 1 fps 720p software encoding in the regular esp32 😄 

[image]