jameszah/ESP32-CAM-Video-Recorder-junior #49: Wlan connection

KiWyFox, 2025-04-23:
i cant connect to my home wlan.
what i noticed over serial monitor is the following:
it only picks up the first word of my ssid.
my ssid contains three words with a space after the first and second word.

Is there a way to fix this in the code?

jameszah, 2025-04-23:
https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ee65f22dc411fcdebf1e5a78e8f742f95b18768b/v62/ESP32-CAM-Video-Recorder-junior-62/ESP32-CAM-Video-Recorder-junior-62.ino#L627

All these lines read the config file until they find a space " " separator.
You could change it to find a tilde "~" or colon ":" separator 

from this:
cname = config_file.readStringUntil(' ');

to this:
cname = config_file.readStringUntil('~');

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ee65f22dc411fcdebf1e5a78e8f742f95b18768b/v62/ESP32-CAM-Video-Recorder-junior-62/config.h#L237

And then the file would change from

```
desklens  // camera name
13  // framesize  13=hd
1800  // length of video in seconds
0  // interval - ms between recording frames 
1  // speedup - multiply framerate 
0  // streamdelay - ms between streaming frames
GMT  // timezone
ssid  // ssid #1 wifi name
12344321  // ssid password
ssid  // ssid $2 wifi name
12344321  // ssid password
ssid  // AP ssid wifi name - switch to camaera name if it is ssid
12344321  // ssid password
```

to 

```
desklens~ // camera name
13~ // framesize  13=hd
1800~
```

etc.

You would probably change the entire file, rather than just ssid.
Might take a little debugging ????

KiWyFox, 2025-04-23:
thanks for your fast reply.

i have replaced the space with a tilde "~" and on second try with  colon ":" as separator.
also edited the config.h with added tilde or colon.
Sadly i still cant connect to the network with spaces in the ssid.

looks like i have no other choice then changing my routers ssid to one without spaces..

maybe you have an idea how to flip the image horizontal and vertical?

[image]
[image]

jameszah, 2025-04-23:
Did you make all the changes ... 7 or 8 lines with read until space?

jameszah, 2025-04-23:
https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior/blob/ee65f22dc411fcdebf1e5a78e8f742f95b18768b/v62/ESP32-CAM-Video-Recorder-junior-62/ESP32-CAM-Video-Recorder-junior-62.ino#L518

jameszah, 2025-04-23:
https://randomnerdtutorials.com/esp32-cam-ov2640-camera-settings/

hmirror and vflip

KiWyFox, 2025-04-23:
My mistake🙈.
I only changed line 627 first. Now that I've changed everything else, I can connect to my home network without any problems. Flipping the image also works perfectly. 
Thank you so much for your help and the great project for saving the ESP32 cam locally on an SD card.